OlegMishin/GotekFlashFloppy #2: First populated Board not working

thbreidenbach, 2022-04-01:
Checked the board the whole day, no sign of life. 
For programming you might consider making 5V accessible on e.g. the uart connector. For USB/serial programming access to 5V is necessary so placing it next to tx would be beneficial.
That said, also debug led might be handy when checking the board.

How did you program the board initially? Is there some clue since UART and USB give no response.
For UART I used 5V from the Drive Connector and connected rx/tx and vice versa and a jumper from Boot0 to 3.3V. No life..
For USB I connected TX/5V and Boot0/3.3V. No connection/Hardware detection possible.

Another question with regards to r4/r5: Are those networks bussed or star?

I'll add some Photos, maybe you can spot an error. For differing footprints I measured for connection. All fine
[image]
[image]
[image]
[image]
[image]
[image]
.

Thanks in Advance!! 

Yours Thorsten

OlegMishin, 2022-04-02:
Hello Thorsten,

Thank you for building my board. The soldering looks very nice. I see no errors.
Regarding the MPU's programming. I program it via SWD interface (as you see, there is X1 connector mounted). I use ST-LINK v2 (cheap Chinese clone) debugger and STM32 ST-LINK Utility SW. Let me update "read.me" file with the short instruction.

Just checked my boards, the DFU mode did not work for me. There is an Errata item in MPU's datasheet, looks like it not possible for earlier releases of the MPU:
[image]
My MPU has "043" date code. Seems this is a problem. If I see correctly, on your MPU the date code starts from 4 - the same issue.

Regarding R4/R5 they are bussed according to Bourns description:
[image]
Please refer to the board's schematics:
[image]

Hope my answers help :)

Regards
Oleg

thbreidenbach, 2022-04-02:
> Hi, was successfully able to program vie uart.
> For easier use 5V should probably be made available next to Tx.
> Drive configuration was an issue since it wouldn’t work nicely alongside another Gotek at first. Drive select etc might be handy.
>
> Will Post success on A1K.org!!
>
>
>
> Thorsten Breidenbach
>
> Von meinem iPhone gesendet

OlegMishin, 2022-04-02:
Thanks for the feedback. The items go to a TODO list.
I have a question about Drive Select. As the selection already implemented via solder-jumpers R6/R7/R10 as on the below picture, do you mean it would be better to have normal jumpers instead?
[image]

thbreidenbach, 2022-04-02:
Hi, indeed solder jumpers are not ideal for an external drive which might easily be used in different systems. Indicator led would be fancy.

Yours
Thorsten

OlegMishin, 2022-04-04:
Hi,
Ok, I will consider at least jumpers or a DIP switch. So it would be possible to change the drive number and see the current configuration without opening the case. I will also update 3D model of the external case.

Regards
Oleg

thbreidenbach, 2022-04-04:
Perfect! Are files for the external case up yet? Would 5V be possible next to TX for chip revisions that offer usb programming? For uart 5V supply is also mandatory and unavailable as of yet. (Labels/Silkscreen would be important)
Yours
Thorsten

thbreidenbach, 2022-05-14:
Hey Oleg, I’m about to order several PCB. Have you had the opportunity to offer some production files? If possible I’ll test as soon as possible. The design looks extremely well.

There was one thing that came to my mind. I use a PC Bridgeboard which needs the JC jumper be set. Since the goldengate provides a db25 con that would be nice.

Hope you’re doing well, your Gotek still works like a charm! Thank you!

Thorsten

OlegMishin, 2022-05-15:
Hi Thorsten,

I'm doing well, thanks. Hope you too.

I'm happy the device is working well for you. Initially it was designed for Amiga use only, but recently I bought an old PC mainboard and started to play with DOS stuff, so it was very useful in a PC mode too. Due to this I added a JC to the DIP switch.
If the design looks ok for you, you can build it using the gerbers I put in "Release 1\gerbers" folder. DRC completed without any issues, so it is safe to build.

I didn't get you point regarding db25, what do you mean? Is it about some other computer connection, other than Amiga/PC?

BR
Oleg

thbreidenbach, 2022-05-15:
Hey Oleg, great News! Will order a bunch of ten immediately.

My point was that the Bridgeboard utilizes the same connection than the Amiga, so it’s an even greater addon with a jc jumper. Some PC Cards have a different external connector I believe and would need different interface. But this fits as is. This way I can use the internal Gotek in my Amiga 4000 as an Amiga Drive. Your board is perfectly fitted for my Big Box Amiga with Bridgeboard!

yours
Thorsten

closed 2025-07-10 by OlegMishin