OlegMishin/GotekFlashFloppy #4: R1 & R2 internal connector PCB

drwhy75, 2022-04-28:
Hi ! 
Nice work !!
I am trying to build one of your Rev0 boards but I am not able to find the values for the Resistors R1 & R2 on the little
internal connector PCB. (could not find / or see them in the BoM).

So far I have finished soldering all components (without R1 & R1 & the buzzer).
... will try to flash the FW when all soldering is finished. 

Cheerz

OlegMishin, 2022-04-28:
Hi!
Thanks! The resistors in fact are jumpers for drive selection. As you can see on the adaptor's schematics, R1 is 0R and R2 is DNP (Do Not Populate). This is a default configuration for drive 0(e.g. internal drive). R2 is for drive 1 configuration (R1 should be removed).

Regards
Oleg

drwhy75, 2022-04-28:
... ahhh ! Thank you !

drwhy75, 2022-04-28:
Hi !

Thank you for answering !

… another question:
What do you think about adding a second connector (DB23 female or just a pin-strip) to the board,
to pass through a 3rd and 4th drive, just like the real external amiga drives where built in the 90´s.

I don’t know if this is possible with the Gotek FW or what other things will be needed for this,
or if its even needed by the AmigaCommunit these days …

Best regards,
Wolfgang

OlegMishin, 2022-04-28:
Hi Wolfgang, you are welcome!

Honestly I did not consider to create something similar to usual Amiga's external drive. The goal of the project is to create a small external floppy emulator, directly plugged-in to the Amiga. I make the adapter pcb to place the device inside of Amiga to replace its internal drive.
Sure, it is possible to connect more than two Goteks with actual FW as the drive selection is a HW thing. There are plenty of old external drives available and my current Gotek+adapter can be installed into them with a 3D printed frame. But in this case it is cheaper to buy an original Gotek and put it into existing case.

Maybe makes sense to develop a classic external drive (with a pass-through connector) completely, including a 3D printed enclosure. But as I'm quite happy with the current one, it could be a long term plan :)

I'm currently collecting feedback for the current release and planning to release the next revision in future. Please have a look at "Revision 1" directory. It contains draft documents for the upcoming pcb. 

Regards
Oleg

drwhy75, 2022-05-04:
... so ... my first board seems to work so far !
(but I did only some short tests, more coming when its really finished soldering !)

I am still waiting for my db23 Connectors and the PiezoBuzzers and, sure I have a wrong sized OLED display too,
so mine looks like that ;-)

[image]

[image]

[image]

[image]

[image]

[image]

but as I said before,
nice project, thank you !

Ah ! One more thing:
I tried to order the "drive-enclosure" via JLCPCB´s 3D printing service but after reviewing the
.STL files they send me an email with the info:

A500_rotary_OLED.STL, CASE_Bottom.STL & CASE_top.STL contains errors and bad edges !
Also the wall-thickness is too thin (only 1mm ... 1.5mm would be better, because of highly risk
of deformation and cracks.

best regards,
Wolfgang

OlegMishin, 2022-05-04:
Hi Wolfgang,

Thanks for your feedback and pictures. Looks very nice :)
I will update the wall thickness to match Jlcpcb requirements.

Regards
Oleg