OlegMishin/GotekFlashFloppy #6: Weird behavior when used as DF1:

nosyn00b, 2023-01-10:
Hi Oleg,
  finally I have mounted a new DF0 floppy, because the old one had not a true ready signal and the hack made a mess, causing the Gotek not to work (typical DF1:???) . Now that I have definitly solved the issue with the internal drive hack I can use your gotek both as a df0 (forcing it as "bootable df1" with my ACA 500), and as df1, in combination with the real df0.

Anyway, when using he real df0 drive and the gotek as df1, I discovered that insertion/change of a disk in the gotek are acknowledged only if the Amiga boots with a disk already inserted from the start. If I boot with a "disk ejected" status on the gotek, there is no way to make the any disk visible/functioning in a second moment: the gotek floppy in this case stays steadly on track 0 afted disk insertion... no way to change that.

Perhaps this is not an obvious behavior, because to have a disk in ejected status from the start you have to configure the gotek to do so (for me is very logical to start  without a disk already interted, so I configured the Gotek in that way). Do you think this is caused by the ZoomFloppy software, or has something to do with the hardware? I suspect that absence of the motor on logics/chips (optional an untested, right?) could have some influence on that, but its only my supposition.

What you think about that? Can you reproduce the same behavior or is something wrong with my gotek?

OlegMishin, 2023-01-14:
Hi,
By chance I have never used my gotek as an external drive without a disk in it. So the issue was not observed.
Truing to reproduce such behavior I did confirm that if i did not insert USB stick and start Workbench, the drive was not detected by system. If I used normal external floppy, booting without any disk in it, the floppy drive was detected by system an reacted on disk insertion. I would assume, it is correct behavior.
Yo are right about optional U4/U5 chips. It is a circuit from real external drive.  It was implemented just in case and was not tested. I checked today and do confirm - with these components, my gotek was detected by system even without USB stick inserted. But if you want to mount them be careful, there is a bug in the circuit. Pins 1,2 of U5 shell be connected to "A-SEL1n" (or 2, 3), not to "A-DIR".

Here is the PCB's mod:
[image]

nosyn00b, 2023-03-16:
Only to let you know that (while searching and finally now waiting) for required components, I had a better read through Keirf documentation about FlashFloppy (FF) and specifically about [Amiga hints](https://github.com/keirf/flashfloppy/wiki/Host-Platforms#commodore-amiga)

From what I understand the behavior is standard with the current FF firmware, and it should be common to  any Gotek without flip-flop circuitry (not only yours when U4 and U5are not installed).

This happens because when the standard “shugart interface mode" is configured, the FF do not send any "Drive ID" on the Ready signal, when motor is off, but simply asserts READY whenever a disk is inserted regardless of the motor on/off state. So this “disk insertion Ready" assertion works as Ready signal when motor is on, and also a a DD “Drive ID-sequence” in case the moto is off. 

This works perfectly when the Gotek is DF0, but when it is used as DF1 with a "passive interface cable" (eg no external enclosure and flip flop motor logic), and the FlashFloppy is configured to eject-at-power-on, or has no USB stick inserted or simply it’s too slow to initialize, then RDY in not asserted during boot when the motor of external drives is enabled and the drive is not managed as it should. 
The good news is that you can “solve” this problem configuring Gotech with the interface = amiga option.
In this case the Ready/Drive ID signal is managed differently from the FF: the Drive ID sequence is constantly sent on the READY pin and makes his work when the motor is off. Because the Binary sequence simulates a Ready signal it also works when the motor is on, so the drives works as expected during boot even if the floppy disk is not inserted from the start.

I tested this configuration and it seems to work even with the “basic” variation.

OlegMishin, 2023-03-18:
Thank you very much for your finding! 
As usual - RTFM :) and I had no time for that :)

closed 2023-03-18 by OlegMishin