OlegMishin/GotekFlashFloppy #7: Setup Rev.1 Question

venice1200, 2023-10-07:
Hi,
this is not an issue, more a question how to setup the board for df1 usage.

I got a Rev.1 PCB and assembled it.
I got FlashFloppy 3.41 installed.
Display, USB and Rotary are working but the board is not detected by my Amiga 500.

Do I only need to close the „jumper“ for drive 1 to activated the board for df1?

[image]

Many thanks for this compact Gotek Drive.

venice1200, 2023-10-07:
Full Board View

[image]

OlegMishin, 2023-10-08:
Hi,
Correct, only DIP "1" shall be ON for Amiga's Drive 1. If there is no issues in soldering it should work. I would recommend to clean the pcb with isopropanol. It looks a bit dirty.

venice1200, 2023-10-13:
It looks like cleaning does the job as I don't find anything wrong.

But is it correct that the LED is slightly blinking all the time even the drive is not accessed?

See:

https://github.com/OlegMishin/GotekFlashFloppy/assets/43762187/e5c33d22-e26a-4733-8af2-41b048c1ded9

OlegMishin, 2023-10-16:
I see the same blinking in my boards but only when Kickstart is about to boot from it. After loading, e.g. AmigaTestKit, no more blinking.

venice1200, 2023-10-16:
The blinking is controlled by the MCU or by the Amiga or from both together?

opengiveback, 2023-10-19:
Hi, congratulations on this great de:sign, and thank you for making it opensource.

I made Revision 1 boards also, and have one assembled and working, but also have a question. Is it possible to use the "Revision 1" PCB based device as DF0: ? I have used a normal gotek before, flashed with FlashFloppy and with an external gotek adapter, and when I set jumper S1, it was DF1:, when I set S0 instead, it became DF0: when I unplug internal floppy, on the external port also.

But **these dip switches start with 1? When I set "1" as on, it acts as DF1: , so how would I get this to be DF0: ?** When the internal drive is not connected, of course.

P.S. I intend to build more and also sell these, and give back 15% of profits to the author(s of hardware and software), as per my opengiveback.org idea. I believe everyone should give back to the authors of products built for sale from open source, so if you have any email for paypal donations, or other means of donating to you preferred by you, please let me know.

venice1200, 2023-10-19:
From my Amiga understanding you can use DF0 only internally without any modification's.

Means, you need the Rev.0 Adapter PCB and set it to Drive 0 to use it internally as DF0, if the Adapter can work with Rev.1. The Board himself should be configurred as Drive 1.

If you want to use DF0 externally you need a switch (Boot selector) which swaps the drive enable signals between DF0 and DF1.

If I am wrong correct me please.

venice1200, 2023-10-19:
Oleg, my issue is solved, feel free to close it.

closed 2023-10-19 by OlegMishin