OlegMishin/GotekFlashFloppy #8: Problems with external version

CultiMAN, 2024-04-18:
Hi Oleg
Thx for a great project.
I have built some of your Goteks. I've always gotten them to work fine for the internal mount.
But now i have tried your external version and I get no active display or activity on the drive.
I'm using your DF0 DF1 switcher and that works fine, but the gotek itself is just dead (tried two)
I see 5 volt on the 5v pins and also 3.3 volt on the display. Is there anything i'm missing to get it to work externally?
[image]
[image]

OlegMishin, 2024-04-19:
Hi @CultiMAN, 
In some boards I had a similar issue. I could flash the MPU via SWD but no startup.  It is related to BOOT0 state at power on. Try to add a 10k resistor between BOOT0 pin at X4 and GND.
Regards
Oleg

CultiMAN, 2024-04-19:
Ok thx. I will certainly try that and get back to you if it worked. I am travelling for work so will test next week.

sbambach, 2024-04-21:
Had same issue... Randomly it booted / or not ;-) 10k pulldown for BOOT0 fixed it.

CultiMAN, 2024-05-01:
Perfect that 10k fix worked! Now though i have another issue. Games work fine when booting from the external Gotek using the adf files already assigned on the usb stick. But when i boot into track 0 to use the menu 
to select more disks i get the error below 
.. Is that something you have seen on the external version? Tried two different usb sticks.... 
[image]

OlegMishin, 2024-05-02:
Sorry, no idea. Never had this issue.
Could you please describe step-by-step how to reproduce it?

CultiMAN, 2024-05-02:
Hi Oleg,
Sure - here is my setup.
1) I use your DF0-DF1 switcher.
2) Switch to use DF1 as boot device.
3) Use the above Gotek (the pictures) connected to the Disk Drive connector.
4) All the game ADF's work fine.
5) BUT when I go to the FlashFloppy menu (ID 0) - I get the error message - so I cannot get to the menu to add game disks to the Gotek. I will try another Gotek build later, to see if I've made some kind of soldering error.

OlegMishin, 2024-05-02:
I got it. The issue is related to "flashfloppy-hxc-file-selector" which I have never used. Usually I just copy ADF files to USB stick and use the knob for navigation.
If I have time I will check how it works with "flashfloppy-hxc-file-selector" and come back.

CultiMAN, 2024-05-02:
An update:
Tried the newest experimental version of FlashFloppy 4.6a. The release notes mentions this error was fixed. But the problem is still the same... I did not fix the issue

OlegMishin, 2025-07-10:
In my understanding it is a software issue, I can't do anything here. It was reported that the last working firmware is 3.29. Maybe a downgrade is the only option for now.