OlegMishin/GotekFlashFloppy #9: Idea for new version

Vanles, 2024-09-27:
Hello,
very nice drive, I like it a lot.

Would it also be possible to install a "5-way switch"? I would really like that and it would look even nicer in terms of design, especially as an external drive. Perhaps it will be possible to take this into account in a new version and install the appropriate soldering points.

Like this one
https://t.ly/8tBXI

Thanks Marco

OlegMishin, 2024-09-27:
Hi Marco,
Thank you for the idea. Typically I use either buttons or an encoder(my favorite) in the Gotek, but can't imagine how to use 5-way switch. To change floppy images only two buttons are needed plus optionally third - "OK".
I wonder, does FlashFloppy SW support 5-way switch?

From design stand point, a low-height encoder's knob also looks good :)

~Oleg

Vanles, 2024-10-22:
Hello,

I think the stick is better than the rotary knob, that's why I asked.

Hi Oleg, 

You could connect left to bottom and right to top and press the selection. That would work well in terms of control.
It would look great. Maybe you can find a solution so that you could also install the stick. 

Maybe even more holes? :)

You could add an "Interactive HTML BOM", the plugin is available for many programs like Egel or KiCat... this makes the assembly easier :)

You're doing a good job with the platinum, I'll be making a donation in the next few days.

Thanks
Marco